ebastler/zmk-designguide #1: suggestion: protection for battery reverse polarity

ergodone, 2021-03-09:
My understanding is there is no protection for a reverse polarity battery connection in the current design.  I think that would be a useful addition since there seems to be no standard for the orientation on the battery connectors (at least the batteries I ordered did not match my adafruit feather).

ebastler, 2021-03-17:
Thanks! That's a good point indeed. Most Schottky Diodes would cost ~.4 V though, which lowers the amount of energy that we can pull from a battery by quite a bit. I'll have to come up with a different idea. 

The charging circuit already has a small flaw (If the board is left for multiple hours/days connected to a charger, with LEDs running, the charger will not correctly determine end of charging, and hold the battery at 4.2V which will damage it over time). Workaround will be implemented.

Percentnineteen, 2022-01-26:
Could use a  channel mosfet between By and GND with B+ as the gate. I think it's done that way in mitosis

Tsangan1, 2025-06-05:
> The charging circuit already has a small flaw (If the board is left for multiple hours/days connected to a charger, with LEDs running, the charger will not correctly determine end of charging, and hold the battery at 4.2V which will damage it over time). Workaround will be implemented.

Hey, I know this was already 4 years ago - but has it been fixed yet? Or would there be and "easy" way too implement it by myself?
 (in the end, any way would be fine, aslong as it fixes it, but of course an "easy" one would be nicer 😄)
I am a little scared about it if I am being honest, so I Just wanted to ask 😅 Thank you!