ebastler/zmk-designguide #2: Question about UG_EN

eigenbrain, 2021-07-02:
Hi Ebastler, 
First of all, thanks for the wonderful material that you are sharing. I am very new to the integrated board design, and your material is of great help. On your schematic, there is UG_EN pin, which controls the underglow circuit. Can you let me know how I set this pin in ZMK firmware configuration?

ebastler, 2021-07-03:
Hi eigenbrain,

glad you like my guide - I'm super proud how many people are using it as a reference for their designs.
I got the UG power toggle working on my isometria 75, mostly by copying the nice!nano - check out the linked line in the *keyboard*.dts file: https://github.com/ebastler/zmk/blob/isometria75-rev2/app/boards/arm/isometria75_rev2/isometria75_rev2.dts#L68

It was on GPIO 1.15, you'd have to replace the <&gpio1 15 GPIO_ACTIVE_HIGH> with <&gpio0 4 GPIO_ACTIVE_HIGH> for the pin chosen in the design guide.

noclew, 2021-08-17:
what a nice guide this is. Thanks for putting this together!

eigenbrain, 2021-08-17:
> Hi eigenbrain,
>
> glad you like my guide - I'm super proud how many people are using it as a reference for their designs.
> I got the UG power toggle working on my isometria 75, mostly by copying the [...]

Thanks for the detailed explanation, and sorry for the late check. Now I am making my first ZMK board following your guide and am so excited. 

By the way, if I end up having the PCB ready, gan you let me know how I can flash the keyboard? I read that you used STLink, but I am too new to guess how it works. If there is any good resource to see how I can make it work like nice!nano, which gets into the bootloader mode when the reset button gets double clicked, can you please point me where to look?

ebastler, 2021-11-30:
Sorry for the late reply - I totally missed the issue you opened.

First of all - good luck with the PCB design! I guess you already finished by now, but I'll address your questions anyway. Better late than never.

1) I was asked about this in another issue today, will link the answer here: https://github.com/ebastler/zmk-designguide/issues/3

2) The behavior you describe is the default Adafruit BL behavior - if you follow jorics guide in my linked explanation, it will work that way.

3) the holyiot docs suggest having the antenna protrude over the PCB edge. I did not have the space necessary to do that, but I guessed adding a hole below the antenna would be better than nothing. I can not tell whether or not it actually helps. Signal strength on the isometria is solid.

Kind regards and sorry once more for the late reply!

closed 2022-10-11 by ebastler