ebastler/zmk-designguide #3: Question about SWDIO/SWDCLK

Forge-Media, 2021-11-30:
I'm looking to build a custom Bluetooth 60% PCB, I've stumbled on your guidelines which are amazing and proving really useful. I've previously designed a working 3x3 macropad following ai03's guide, so have some experience, but introducing Bluetooth, a battery and zmk is all new.

I was initially going to just use a Nice!nano v2. But since reading your guidelines I'm hoping to do it all myself, using a Holyiot 18010.

> Under the MCU section regarding SWDIO/SWDCLK these must be wired to contact pads, because this is where the STlink/Jlink goes for flashing.

This part has me a little confused. I've not used zmk before, but from reading the docs flashing sounded quite simple,  place board into bootloader mode (double press the reset switch) and drop the UF2 file onto it. However, from reading your guidelines it sounds that there may need to be more work involved to get the Holyiot 18010 flashed before flashing zmk? 

- Does the Holyiot 18010 come without a pre-loaded boot loader?
- Could you go into any detail about what is involved with STlink/Jlink for flashing or are there any other references you could link me to?

Thanks

ebastler, 2021-11-30:
Hi,
Sadly all nRF52 modules I know of (apart from maybe adafruit ones?) come without any bootloader, and unlike some stm32 chips the nRF52 series does not have a in-built bootloader in ROM.

Once the bootloader is on, flashing is - as you pointed out - super simple and can be done via USB UF2 drag and drop. In order to get the bootloader on the module, you need either a STLink (works only for unlocked modules, holyiots appear to always be unlocked) or a Jlink (works for locked or unlocked modules).

For flashing you need a computer running the openOCD tool (i have only used it on Linux so far, not sure if there is a windows/mac version of it), and the flashing itself boils down to 1-2 commands, depending on whether the module has a locked or unlocked BL (only 1 for holyiot).

Joric wrote a good wiki article regarding flashing, i linked you the stlink section. Jlink is right below. You only need the "flashing" command on a 18010, not the "unlocking" command.
https://github.com/joric/nrfmicro/wiki/Bootloader#st-linkv2

Hope I could help!
Moritz

Forge-Media, 2021-11-30:
Thanks, will take a look at the link.

Flashing the bootloader still seems to be the one thing I'm most nervous to tackle. I'm hoping any equipment I need won't be too costly.

 do you think it's possible to have the Holyiot 18010 supplied with the Adafruit nRF52 Bootloader pre flashed or have someone like PCBway flash it for you if I'm paying for assembly?

ebastler, 2021-11-30:
Hi! A Jlink will cost you ~20$, a STlink v2 Clone used to be 1-2$ on Aliexpress, nowadays they are around 3-4$ as far as i know (chip shortage...). Still very affordable and all you need. The most difficult part is to install openOCD (you have to compile it from source on Linux), but that boils down to ~5 commands I can dig up from my chat history from the last time I explained it to someone if you should need it - no clue how much experience with Linux you have.

Usually PCB fabs offer flashing for assembled boards, but I have yet to try if that works for nRF chips too. I guess it should.

Forge-Media, 2021-11-30:
Thanks, that would be highly appreciated. I've a fair amount of Linux experience with servers, though all cloud and virtual. Don't have a dedicated Linux machine at home apart from a Raspberry pi 4, which seem could be an option with OpenOCD according to the link you provided and I won't need to purchase anything else.

I'm not really sure I trust the assemblers to flash it more than myself. 😅 But the cost for the equipment is not really an issue, was more worried it would be much higher and more difficult to find in stock.

ebastler, 2021-11-30:
I tried getting openOCD to run directly on a Pi3b, but failed. Not sure if I missed something. Sadly docs for how to configure openOCD directly on a Pi are sparse. I ended up using the STlink v2 connected to the Pi via USB - that worked fine.

Forge-Media, 2021-11-30:
Ok that's good to know, I may order an STlink v2 in that case.

Forge-Media, 2021-11-30:
One last question, did you purchase the Holyiot 18010 directly from the manufacturer and did you you hand solder it to your PCBs or have them soldered as part of a PCB assembly service?

ebastler, 2021-11-30:
I ordered mine off Aliexpress, I think it was their official store. As long as you only use the outer pads, they are super easy to hand solder.

Forge-Media, 2021-12-01:
I contact Holyiot via email to ask about the bootloader and they told me the following: 

> Yes, we can pre-program your hex file before shipping.
> We provide this service for free now and in future.
> Please place the order and leave the message on Aliexpress at the same time.

I'm thinking this may be a good idea. But thought I would ask which bootloader version you used. I'm assuming there should not be any issue sending them the latest; [pca10059_bootloader-0.6.2.zip](https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases/tag/0.6.2)

I did read on the ZMK discord I could build my own version of the Adafruit bootloader, and flash that. This was suggested by a user called petejohanson and he linked to his custom bootloader: https://github.com/petejohanson/Adafruit_nRF52_Bootloader/tree/technikable/src/boards/technikable 

Besides specifying the pin of the the user LED, and showing a custom mass storage device name to the OS. I'm not sure if there is an advantage to building my own bootloader versus using the pca10059_bootloader-0.6.2.

ebastler, 2021-12-01:
Okay, that is super cool. Didn't know they offered this service. I'd go for it.
I think I just used the default adafruit one for my isometria, can't really remember though to be honest. Pete is the project lead of ZMK, so he sure knows what he's doing, but without any deeper research I don't really see the advantage in doing so either.

closed 2022-10-11 by ebastler