ebastler/zmk-designguide #4: Question about battery connection.

BeefcakeTheMighty, 2021-12-10:
Thank you for all of the work you've put into this project and for sharing it; it's awesome!

I'm not understanding how the battery powers the MCU. I see VBAT out on the battery management and VBAT in on the voltage reg., VSENSE, and underglow. It appears that the only power the MCU receives is directly from USB-D+/-.

Also, I'm not seeing what VBUS on the USB-C connector supplies. 

Maybe I'm just missing something super simple; any guidance you can provide is much appreciated!

ebastler, 2021-12-11:
You're welcome - glad you like it!
Regarding your first question - the MCU can't take the full VBAT, hence VBAT is passed through the voltage reglator (the section between battery management and the MCU in the guide), which outputs stable 3.3V for the MCU (module).
The 5V pin on the MCU is only used as a USB plug detection (and I think powers the USB peripheral), but not the MCU itself. That one is powered by the 3.3V on pin14.

VBUS on the USB-C connector is the 5V pin of the connector, called VBUS since it's the supply voltage delivered by the USB bus - I call it 5V after the fuse.

Hope that answers your questions - if I missed a point, feel free to ask :)

Cheers,
ebastler

BeefcakeTheMighty, 2021-12-11:
Oooohh...yup...I see it now. I was focusing too much on the "direction" of the wires and should have paid more attention to the nrf documentation. 

I think you covered all my questions! Thank you!

The only one I have remaining is:

Are all the 5V+ tied to the same "rail"? I.E. the 5v/VBUS supply on the usb-c (through U2), the two on the battery mgmt. (assuming I calculate my own R14 & R16 and use a stand-alone thermistor), and the 5v pin/VBUS on the MCU (module)?

Thanks in advance!

ebastler, 2022-10-11:
Hi! I am terribly sorry, I must have missed this comment - only saw it now, almost a year later. Guess you don't need it anymore, but yes, all +5V labeled nets are supposed to be a single net!

closed 2022-10-11 by ebastler