ebastler/zmk-designguide #7: Ask: Please provide the ZMK config

LamDVT, 2022-12-19:
Thank you for your project. It's awesome!
I used your guide and it worked correctly.
Can you provide your ZMK config (used to run the hardware in the tutorial) for us to refer to and follow?
Thank you very much!

ebastler, 2022-12-19:
For the firmware:
https://github.com/ebastler/zmk-config/tree/main/config/boards/arm

Bootloader:
https://github.com/ebastler/Adafruit_nRF52_Bootloader/tree/ebastler/src/boards

altar is a holyiot + voltage divider based board

wlpad is a Moko + Fuel gauge based board

Both use bq24075, but that doesn't really affect the code.

doantruonglam, 2022-12-20:
Thanks forThanks for your quick answer.
I'm having problems with UG_EN can't activate UG_PWR for ws2812. I followed the instructions here: https://zmk.dev/docs/config/power but failed, I would like to refer to your source code to configure this section. 
Thank you.

ebastler, 2022-12-20:
Oh, that's the issue - check my Isometria 75 code. That one is using a Holyiot with UG_PWR toggle!

doantruonglam, 2022-12-20:
Your answer is very helpful with me!
Last question: I do not need care about the line `ALERT ` in "Voltage sensing (advanced)", right? Can I use both `MAX17048` and an OLED I2C by connect to the same `SCL, SDA` lines?

ebastler, 2022-12-20:
I didn't use the Alert yet, the driver doesn't make use of it either I think. I have not tried OLED + Chip, but shouldn't be an issue - i2c is made for multiple devices on the same bus.

Just one thing: the driver for the fuel gauge is not officially merged into ZMK yet. 

I have merged the driver and patched an issue with it on this branch of my fork: https://github.com/ebastler/zmk/tree/testbranch

You can use that, I will try to set the PR in motion these days, so hopefully it will sonish be working with the main ZMK.

doantruonglam, 2022-12-21:
I'll give it a try. I think it should work properly.
Thank you for your support.

ebastler, 2022-12-29:
No problem, hope you get your design working!

closed 2022-12-29 by ebastler