ebastler/zmk-designguide #11: holyiot 18010 uf2 bootloader

michaelrommel, 2024-01-21:
Hallo Moritz,

entschuldige diese Art der Kontaktaufnahme, leider schlugen alle Varianten, Dich auf Discord oder twitter zu kontaktieren fehl.

Ich habe im Mosfetkiller Forum gelesen, dass Du mal eine Tastatur mit dem Holyiot 18010 Modul gebaut hast. Ich hab das auch gemacht, aber ich bekomme den Adafruit UF2 Bootloader einfach nicht zum connecten, wenn ich den Reset Taster doppelklicke, passiert ausser dem Reset (sehe ich auch nur an der Stromaufnahme) gar nix.

Ich bin jetzt etwas ratlos und suche nach jemandem, der den Bootloader zum Laufen mit dem Modul gebracht hat un dmir evtl. sagen kann, ob man ihn neu kompilieren muss und welche settings entscheidend sind, z.B. hat ja das holyiot Modul die DCDC Wandler aktiviert etc. 

Hast Du hier vielleicht einen Tipp fuer mich oder ein repo, in dem ich mir Deine Loesung ansehen koennte?

Vielen Dank im Voraus!

  Michael.

ebastler, 2024-01-21:
Hi! 

Kein Problem, passt hier ja auch ganz gut :)

Allem voran - hast du den Bootloader korrekt geflasht gekriegt? Also mit post-flash-check ohne Fehler? Klingt für mich in erster Linie als wäre der Bootloader gar nie drauf.

Die nRF kommen ab Werk mit Write Protection, vergessen viele die erstmal aufzumachen, dann schreibt der Flash nichts. Welchen Programmieradapter nutzt du?

An sich braucht das Modul nichts besonderes - ob man DC-DC beim Adafruit Zeug in der FW oder im Bootloader aktivieren muss erinnere ich mich nicht mehr, standardmäßig sollte es aber auf aus stehen. Welchen Bootloader hast du genau genommen? Selber einen zusammengestellt, oder einen der Sample BL verwendet?

Ich könnte Mal einen neuen Workflow in meinem Repo anstoßen, der Bootlader der Isometria war für ein Holyiot ohne irgendwas Besonderes ausgelegt. Dann hättest du wenigstens Mal eine BL Config die garantiert auf einem Holy lief zum Testen um eine Fehlerquelle auszuschließen.

michaelrommel, 2024-01-22:
Hi Moritz,

erstmal vielen Dank für die prompte Reaktion!

Zum Testen habe ich verschiedene Bootlader aus dem Adafruit Repo unter Releses probiert, habe keinen selbst kompiliert (aus macos selbst mit brew installerter Toolchain gab es Fehler, ich wollte als Nächstes mal einen BL auf Linux probieren zu kompilieren). Ich hatte recherchiert, dass der 10056 eigentlich haette funktionieren sollen.

Als Programmieradapter habe ich die Gpio Pins eines Raspberry Pis mit openocd genommen, damit hatte ich schonmal ein XIAO BLE Modul neu geflashed, wusste also, dass das prinzipiell funktionieren sollte. Das 'verify" kam auch mit "ok" zurueck.

sudo openocd -f interface/raspberrypi-native.cfg -c “transport select swd” -f target/nrf52.cfg -c init -c “reset init” -c halt -c “nrf5 mass_erase” -c “program Seeed_XIAO_nRF52840_Sense_bootloader-0.6.1_s140_7.3.0.hex verify” -c reset -c exit

Die Tastatur ist hier, (https://github.com/michaelrommel/nightliner), habe die linke Haelfte mal teilw. reflowed, alle SMD Komponenten auf der Vorderseite sind drauf. (OK, zwei habe ich wieder entfernt: hatte die Pinbelegung des USBLC6 falsch und habe den abgeloetet und ueberbrueckt, ebenso wie die Schottky Diode, weil ich ausschliessen wollte, dass der Voltage Drop auf ca. 4.8-4.9V nicht die Ursache des USB Problems ist.)

Von der von DIr genannten Write Protection hatte ich auch gelesen, weiss aber nicht, ob ich da die richtigen Kommandos via openocd gesendet habe. 

dap apreg 1 0x04 0x01

Da der openocd connected und auch die Status Zeile "Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints" bringt, hoffte ich, dass dies erfolgreich war.

Wenn Du bestaetigst, dass eigentlich ein Standard BL mit dem Holy laufen sollte, bin ich geneigt, den Fehler auf meiner Platine zu sehen. Ich bin momentan unterwegs und habe keinenZugang zur Reflow Station und den weiteren Platinen, das Experiment "teilbestuecke eine Platine nur mit USB Socket, Voltage Regulator und Holyiot (+2x 5.1k und ein paar Caps)" muss also bis Freitag warten... Ich hoffte es sei ein SW Config Problem, die Tastatur und den Pi plus Macbook habe ich dabei...

Wenn Du wirklich ein Minimalworkflow fuer den Holy baust, waere das absolut super, dann kann ich mich daran lang hangeln...

Besten Dank und Grüße nach Berlin!

  Michael

michaelrommel, 2024-01-27:
Hi Moritz,

meine Tastatur laeuft nun grundlegend! Der Fehler lag daran, dass ich - nachdem ich den falsch verdrahteten USBLC6 ausgeloetet und ueberbrueckt hatte - das USB-C Receptacle nicht mehr sauber eingeloetet hatte. Die Stromversorgungs Pins hatten noch Kontakt aber die Datenpins nicht mehr. Mir ist das nicht aufgefallen, weil ich die Kontakte immer nur an den Pads gemessen hatte und nicht am anderen Ende des USB Kabels. Muss mir mal ein Breakout Board bauen, das die USB-C Leitungen testbar macht... Heute hatte ich mal versucht ein USB-C zu USB-C Kabel an einem Rechner zu nehmen und hatte keinen Strom (weil er die 5.1k Pulldowns nicht gesehen hat). Das kam mir dann spanisch vor und nun habe ich das komplette USB-C Receptacle getauscht und sehe nun die UF2 Partition. Puh! Ich hab mir die Haare ausgerauft, und dann nur eine kalte Loetstelle.

Aber ich kann nun bestaetigen, dass der Standard pca10056_bootloader-0.8.2_s140_6.1.1.hex Adafruit Bootloader ohne Modifikation mit dem Holy funktioniert! Vielen Dank fuer Deine Troubleshooting Tipps und fuer osprey, da habe ich heute nochmal Analogien zu meinem Board gezogen, da doch einige Bauteile gleich sind, war das eine super Referenz! Muss noch die Abweichungen versuchen zu verstehen: Du hast zwischen Shield/Earth und GND 100nF (ich nur 1nF) und dann hast Du noch sowohl zwischen GND und GND parallel zur TVS Diode BSD3C051V eine Spule BLM15PX121SN1D, und die ist auch noch inline im VBUS. Das muss ich mir mal in den Datenblaettern ansehen...

Jedenfalls kann mein Tastaturprojekt nun weitergehen... Lieben Dank nochmals,

  Michael.

closed 2024-01-27 by michaelrommel