ebastler/zmk-designguide #12: Where do GNDPWR and Earth connect to?

xpol, 2024-09-16:
Hi,

I'm a software engineer, Thank you for the guide!

I have a dumb question: In the "USB connectors and protections" section, there are GNDPWR and Earth besides GND.
So where do GNDPWR and Earth connect to?
[image]

ebastler, 2024-09-16:
Ugh that is a leftover from me overhauling the entire ground/shield handling. Sorry. GNDPWR should be regular GND. Earth is just a label and can stay there as-is. Earth should be what your enclosure is connected to, if the PCB gets put into a metal housing.

krikun98, 2024-09-16:
So is the old schematic with the 1M resistor from the Osprey no longer relevant?