ebastler/zmk-designguide #16: VSW?

vinniefranco, 2025-01-29:
First, thanks for this wonderful guide!

That said, I'm a bit thrown by the underglow portion of the guide. +VSW is labeled but not called out in the docs. What is this connection in the context of a WS2812C context where the level shifter is omitted? Thanks

[image]

ebastler, 2025-01-31:
Hi! +VSW is basically the switched battery voltage, as seen in section https://github.com/ebastler/zmk-designguide?tab=readme-ov-file#simple-implementation and https://github.com/ebastler/zmk-designguide?tab=readme-ov-file#advanced-implementation

In one case it is after the MOSFET ideal diode, in the other it is the system power output of the more advanced TI battery management chip

vinniefranco, 2025-02-04:
Thanks so much!

closed 2025-04-01 by vinniefranco