ebastler/zmk-designguide #17: Question about voltage of ws2812

Gasiro, 2025-03-21:
[image]

Hi ebastler,

Thank you for this guide! It has been incredibly helpful, especially since I had very little experience with schematic design before.

I have a question regarding the WS2812 LEDs. I’m not quite sure about the UG_PWR label—what power source should I connect this to? In most keyboard designs, I assume it would be tied to +5V, but in this case, would you recommend +5V, VSW, or perhaps +3.3V?

Also, I’m a bit confused about the +3.3V rail in the schematic. Where does it originate? Does it come from the VDD pin of the E73 MCU?

I’m still a beginner when it comes to schematic design, so I hope my questions aren’t too silly. Thanks in advance for your help!

ebastler, 2025-04-13:
Hi!

No worries about the questions. The power supply scheme on these boards is quite confusing after all :D Too many separate voltage domains and too many chips and parts involved.

The different voltage domains on this guide all have different reasons. I'll try to explain it a little. Not sure if it is clear the way I explain. Feel free to ask in more detail if you need!

+5V is the power coming from USB. It is always 5V.

+VSW is the power coming from battery management, and will be from battery (if battery powered) or from USB (if plugged in). +VSW can be about the entire battery voltage range plus a bit of tracking voltage from the charging chip - so ~ 3V to 4.5V depending on battery charge level.

VDD is only used as IO voltage for the MCU itself (and, in the example you listed, generated by the MCU itself as well). All signals cominmg from the MCU will be at VDD (3.3V with default setup, I tend to run mine a bit lower at 2.8V for slightly better battery life). The nRF52 has a built-in DC-DC converter that creates the 3.3V it needs (and also 1.8V which it also needs internally) from +VSW. 

UG_LV and UG_HV is a signal level shifter. It lifts the RGB data signal from VDD up to VSW, since 3.3V signal would not be enough data voltage for the WS2812 to reliably work within specs. This chip has nothing to do with power/supplies, it only boosts the data line.

UG_PWR is the same as +VSW, but it is switched on/off by Q3. WS2812 consume quite a lot of power even when they are turned off, so they would chug the battery empty in no time. Therefore, Q3 turns the WS2812 off when not in use. You don't connect anything else to UG_PWR - it is supplied through Q3, and the LEDs use the power. 

So to sum it up, the WS2812 run at the same voltage as +VSW (but with Q3 as a power switch in line with the supply), and the data signal (UG_HV) is at the same voltage as +VSW as well.

Gasiro, 2025-05-07:
Thank you so much — your explanation was super clear, and I now understand everything much better!

Turns out the issue that had me stuck for several days was indeed (as always!) a voltage problem 😅 I was using the Adafruit firmware for flashing, but it didn’t set UICR_REGOUT0 to VOUT_3V3, so the VDD was stuck at 1.8V. The keyboard was technically working, but of course, none of the LEDs lit up. 😆

Now that I’ve fixed that, everything is running smoothly. Really appreciate your designguide! 🙏

[image]

ebastler, 2025-05-08:
Glad to hear it is working now! I see, this can be tricky to debug...

With 1.8V signal the 2N7002 for the highside switch will most likely not work yet, so the power never gets turned on for the LEDs.

Enjoy your board!

closed 2025-05-08 by ebastler