ebastler/zmk-designguide #18: Question about ebyte IC

vinniefranco, 2025-04-11:
Hello!

Am I correct in the assertion that there is no LDO to get 3V3 and it comes from C19 out from the IC?

ebastler, 2025-04-13:
Hi! Yup. The nRF52840 chip has a built-in DC-DC voltage regulator to create the 3.3V VDD supply it needs. L3 and C12 are essential for this DC-DC converter as well, and need to be placed (very) close to pins 25 and 19 as well.

vinniefranco, 2025-04-13:
Thank you! and sorry, for not using the same references as the project! In my schematic the references are L5 and C19

When you say very close, is this sufficient?

[image]

Thank you!

ebastler, 2025-04-13:
This is plenty close! Looks great.

closed 2025-04-13 by vinniefranco